easybotics/node-red-contrib-easybotics-ina219 #2: "Error: EMFILE: too many open files, open '/dev/i2c-1'"

Noodleyman, 2019-06-03:
I left a test flow running all night, and this morning noted the following error from the INA219 node.

It is configured to poll every 500ms.

the failure killed the flow entirely.

keptan, 2019-06-03:
okay /dev/i2c is being opened over and over by inaInit without being closed
easiest fix is probably patching the underlying ina library

Noodleyman, 2019-06-04:
I'll be happy to test out any code changes, or updates. just let me know the steps you want validated. :)

keptan, 2019-06-05:
https://github.com/easybotics/node-red-contrib-easybotics-ina219/commit/c137a641790527a0247d97b404b23ae91f9d9bc7

should clear up this issue, hope to hear from you if it worked or not

Noodleyman, 2019-06-05:
I will set it back up and leave it running for a while to confirm. thank you :)

Noodleyman, 2019-06-05:
Confirmed fixed.

Before the fix:
`cat /proc/sys/fs/file-nr`

the output would keep increasing until we hit the limit.

after the fix:
the output holds steady and works correctly

I think this change is good to go :)

Thank you!

closed 2019-06-05 by keptan