easybotics/node-red-contrib-easybotics-ina219 #3: INA219 node outputs two readings per output instead of one

Noodleyman, 2019-06-03:
Testing for a flow where all I require is the current, however each time the lower output sends data, it sends two readings back to back instead of a single one.

Note, the two readings are not duplicates (in some cases) as the values can vary slightly. there are two readings instead of one being taken and sent within milliseconds of each other.

Set to output the mA every 5 seconds, debug direct from the mA output:
[image]

Noodleyman, 2019-06-03:
This fixed it for my application...

```
                node.vOutput = function (voltage)
                {
                        v = voltage
                        if(v == undefined || a == undefined) return

                        const msg0 = {payload: v, topic: 'voltage'}
                        //const msg1 = {payload: a, topic: 'miliamps'}
                        const msg1 = null
                        node.send([msg0, msg1])
                }

                node.aOutput = function (amps)
                {
                        a = amps
                        if(v == undefined || a == undefined) return

                        //const msg0 = {payload: v, topic: 'voltage'}
                        msg0 = null
                        const msg1 = {payload: a, topic: 'miliamps'}
                        node.send([msg0, msg1])
                }

```

keptan, 2019-06-03:
Yeah you're on the right track, might use your patch here.

keptan, 2019-06-05:
https://github.com/easybotics/node-red-contrib-easybotics-ina219/commit/fa27507f45a2c9cc6e19dfd8b63776c88bac94ac

fixed this bug

closed 2019-06-05 by keptan