easybotics/node-red-contrib-easybotics-ina219 #6: Unable to install

Noodleyman, 2019-10-12:
Using latest node-red release with rPi and trying to setup on a new device.

Everything installs successfully, however after you setup the INA219 node and deploy the following error is thrown:

Flows stopped due to missing node types.
ina-sensor-manager
ina-sensor

clowrey, 2019-10-12:
Hmm we will look into this - it has not been tested with the latest version I believe. Can you specify which exact version of raspbian and which exact version of node red you are running so we can try to duplicate?

keptan, 2019-10-13:
looks like a broken dependency
it should be installing this: https://github.com/easybotics/node-ina219 library instead of the base node-ina219 as a dependency, because we added the `calbreate32V2AResistor` function

it looks like I might have updated the npm package but not the github repo

keptan, 2019-10-13:
https://github.com/easybotics/node-red-contrib-easybotics-ina219/commit/f03780773e60d2dcbf67a4501a6058ac0978ebf0

updated the github repo, so if you install now it should work

closed 2019-10-13 by keptan

Noodleyman, 2019-10-14:
Morning,

Thank you! I managed to fix it shortly after adding this issue via a manual install. I think I made a rookie error with the install and wasn't in the correct directory when running the install command.

Might be worth adding a step to the NPM install notes to change directory prior to installing via NPM to:
$HOME/.node-red

might help save somebody some time if they run into similar issues in the future :)