easybotics/node-red-contrib-easybotics-ina219 #9: Wrong voltage calculation

Luc3as, 2020-04-09:
Hi, I need your help, 
I have very basic setup, wiring is correct, I found old closed issue about common ground. 
I have power supply around 24 volts, 3 A, my module should be capable of measuring up to 3.2 A by the datasheet. but somehow if the voltage is bigger than cca 9 volts, I start to receive negative values. 
[image]
I tried the very basic python sketch and this looks great so there has to be someting in the js library, but I could't find any problem. 
[image]

I also noticed there is no change in voltage measuring if I change resistor value, in settings. 
Don't you have any clue what could be wrong ? thanks

keptan, 2020-04-10:
Thanks for your well documented issue, I'll check this out.

TechMindsYT, 2023-02-16:
Was this issue ever fixed? With 18v in I see negative 15 on the node output. :-)