easybotics/node-red-contrib-easybotics-ina219 #13: Wrong readings when using two ina219 at the same time

Jamos1988, 2021-01-27:
I hope I am at the right address! I am having a problem with the node-red node for ina219.

When using two ina219 with unique i2c addresses (0x40 and 0x41 for the other one) and using a flow in node-red, the readings start to become wrong. When I deactivate one of the flows (for example the 0x40 one) the 0x41 one will start to report correct readings and also the other way around. But with both of them together they get corrupted.

This is the flow I am using

`[{"id":"c802d97a.995758","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"4a1d7508.60855c","type":"tab","label":"12v","disabled":false,"info":""},{"id":"141c5a71.8ef856","type":"tab","label":"UPS","disabled":true,"info":""},{"id":"8715a7c2.f768b8","type":"ui_tab","name":"Home","icon":"dashboard","disabled":false,"hidden":false},{"id":"42d595d.d0b4a6c","type":"ui_base","theme":{"name":"theme-light","lightTheme":{"default":"#0094CE","baseColor":"#0094CE","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif","edited":false},"darkTheme":{"default":"#097479","baseColor":"#097479","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif","edited":false},"customTheme":{"name":"Untitled Theme 1","default":"#4B7930","baseColor":"#4B7930","baseFont":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif"},"themeState":{"base-color":{"default":"#0094CE","value":"#0094CE","edited":false},"page-titlebar-backgroundColor":{"value":"#0094CE","edited":false},"page-backgroundColor":{"value":"#fafafa","edited":false},"page-sidebar-backgroundColor":{"value":"#ffffff","edited":false},"group-textColor":{"value":"#1bbfff","edited":false},"group-borderColor":{"value":"#ffffff","edited":false},"group-backgroundColor":{"value":"#ffffff","edited":false},"widget-textColor":{"value":"#111111","edited":false},"widget-backgroundColor":{"value":"#0094ce","edited":false},"widget-borderColor":{"value":"#ffffff","edited":false},"base-font":{"value":"-apple-system,BlinkMacSystemFont,Segoe UI,Roboto,Oxygen-Sans,Ubuntu,Cantarell,Helvetica Neue,sans-serif"}},"angularTheme":{"primary":"indigo","accents":"blue","warn":"red","background":"grey"}},"site":{"name":"Node-RED Dashboard","hideToolbar":"false","allowSwipe":"false","lockMenu":"false","allowTempTheme":"true","dateFormat":"DD/MM/YYYY","sizes":{"sx":48,"sy":48,"gx":6,"gy":6,"cx":6,"cy":6,"px":0,"py":0}}},{"id":"ccbcb301.5f66f","type":"ui_group","name":"Dashboard","tab":"8715a7c2.f768b8","order":1,"disp":true,"width":"6","collapse":false},{"id":"96e564d9.a0f6c8","type":"ina-sensor-manager","address":"0x41","delay":"1000","ohms":"0.01","customResistor":false},{"id":"5072781c.9046f8","type":"ina-sensor-manager","address":"0x40","delay":"1000","ohms":"0.01","customResistor":true},{"id":"8da05ed4.9f81f","type":"rpi-gpio out","z":"c802d97a.995758","name":"","pin":"11","set":true,"level":"0","freq":"","out":"out","x":440,"y":80,"wires":[]},{"id":"81c1cd7c.62e9d","type":"rpi-gpio out","z":"c802d97a.995758","name":"","pin":"12","set":true,"level":"0","freq":"","out":"out","x":440,"y":132,"wires":[]},{"id":"e93b7190.f6715","type":"rpi-gpio out","z":"c802d97a.995758","name":"","pin":"13","set":true,"level":"0","freq":"","out":"out","x":440,"y":182,"wires":[]},{"id":"7a2b0e26.28e2","type":"rpi-gpio out","z":"c802d97a.995758","name":"","pin":"15","set":true,"level":"0","freq":"","out":"out","x":440,"y":234,"wires":[]},{"id":"1fcb30e2.36d20f","type":"ui_switch","z":"c802d97a.995758","name":"","label":"switch","tooltip":"","group":"ccbcb301.5f66f","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"str","onicon":"","oncolor":"","offvalue":"false","offvalueType":"str","officon":"","offcolor":"","x":150,"y":80,"wires":[["8da05ed4.9f81f"]]},{"id":"5d9fe981.6e7708","type":"signalk-send-pathvalue","z":"4a1d7508.60855c","name":"","source":"","x":750,"y":140,"wires":[]},{"id":"f3918cff.9c8ab8","type":"function","z":"4a1d7508.60855c","name":"current&power","func":"msg.topic = \"electrical.batteries.system.current\";\nmsg.payload = Number(msg.payload)/1000;\nnode.send(msg);\nmsg.topic = \"electrical.batteries.system.power\";\nmsg.payload = Number(flow.get('voltage'))*Number(msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":200,"wires":[["5d9fe981.6e7708"]]},{"id":"4ac9724a.23dc94","type":"function","z":"4a1d7508.60855c","name":"voltage","func":"msg.topic = \"electrical.batteries.system.voltage\";\nflow.set('voltage', msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":420,"y":140,"wires":[["5d9fe981.6e7708","d116b1fc.857e3"]]},{"id":"9dd53318.b65b58","type":"ina-sensor","z":"4a1d7508.60855c","name":"","handle":"5072781c.9046f8","x":230,"y":140,"wires":[["4ac9724a.23dc94","8952c348.335dd"],["f3918cff.9c8ab8","8952c348.335dd"]]},{"id":"14da9dcd.96c132","type":"signalk-send-pathvalue","z":"141c5a71.8ef856","name":"","source":"","x":710,"y":80,"wires":[]},{"id":"757618d3.9ae188","type":"function","z":"141c5a71.8ef856","name":"current&power","func":"msg.topic = \"electrical.batteries.rpi.current\";\nmsg.payload = Number(msg.payload)/1000;\nnode.send(msg);\nmsg.topic = \"electrical.batteries.ups.power\";\nmsg.payload = Number(flow.get('voltage'))*Number(msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":420,"y":140,"wires":[["14da9dcd.96c132"]]},{"id":"fc2f94d2.4c5b58","type":"function","z":"141c5a71.8ef856","name":"voltage","func":"msg.topic = \"electrical.batteries.ups.voltage\";\nflow.set('voltage', msg.payload);\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":400,"y":80,"wires":[["14da9dcd.96c132","5453d74d.85ffd8"]]},{"id":"55528798.f955b8","type":"ina-sensor","z":"141c5a71.8ef856","name":"","handle":"96e564d9.a0f6c8","x":200,"y":120,"wires":[["fc2f94d2.4c5b58"],["757618d3.9ae188"]]},{"id":"d116b1fc.857e3","type":"ui_gauge","z":"4a1d7508.60855c","name":"","group":"ccbcb301.5f66f","order":1,"width":0,"height":0,"gtype":"gage","title":"System voltage","label":"volts","format":"{{value}}","min":0,"max":"15","colors":["#00b500","#e6e600","#ca3838"],"seg1":"","seg2":"","x":730,"y":60,"wires":[]},{"id":"5453d74d.85ffd8","type":"ui_gauge","z":"141c5a71.8ef856","name":"","group":"ccbcb301.5f66f","order":2,"width":0,"height":0,"gtype":"gage","title":"UPS voltage","label":"volts","format":"{{value}}","min":0,"max":"15","colors":["#00b500","#e6e600","#ca3838"],"seg1":"","seg2":"","x":700,"y":40,"wires":[]},{"id":"4b728ee9.6ee97","type":"ui_switch","z":"c802d97a.995758","name":"","label":"switch","tooltip":"","group":"ccbcb301.5f66f","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"str","onicon":"","oncolor":"","offvalue":"false","offvalueType":"str","officon":"","offcolor":"","x":150,"y":120,"wires":[["81c1cd7c.62e9d"]]},{"id":"4fa3548e.82336c","type":"ui_switch","z":"c802d97a.995758","name":"","label":"switch","tooltip":"","group":"ccbcb301.5f66f","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"str","onicon":"","oncolor":"","offvalue":"false","offvalueType":"str","officon":"","offcolor":"","x":150,"y":160,"wires":[["e93b7190.f6715"]]},{"id":"4ce80f4.859d6f","type":"ui_switch","z":"c802d97a.995758","name":"","label":"switch","tooltip":"","group":"ccbcb301.5f66f","order":0,"width":0,"height":0,"passthru":true,"decouple":"false","topic":"","style":"","onvalue":"true","onvalueType":"str","onicon":"","oncolor":"","offvalue":"false","offvalueType":"str","officon":"","offcolor":"","x":150,"y":200,"wires":[["7a2b0e26.28e2"]]},{"id":"8952c348.335dd","type":"debug","z":"4a1d7508.60855c","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":410,"y":280,"wires":[]}]`

Hope you can help!

ryanmich251, 2021-01-29:
Could you try using the address 0x40 for one and 0x42 for the other? If I recall correctly, the last bit in an i2c address is the data direction bit, so 0x41 would actually be the same address as 0x40, just with the data direction bit set (but probably ignored by the library).

Jamos1988, 2021-01-29:
I'll try, thanks!
Just one question though, as Adafruit describes on their webpage:

Up to 4 boards may be connected. Addressing is as follows:
Board 0: Address = 0x40 Offset = binary 00000 (no jumpers required)
Board 1: Address = 0x41 Offset = binary 00001 (bridge A0 as in the photo above)
Board 2: Address = 0x44 Offset = binary 00100 (bridge A1)
Board 3: Address = 0x45 Offset = binary 00101 (bridge A0 & A1)

So, it seems that it is possible that two boards can have an address that is just one decimal apart.
https://learn.adafruit.com/adafruit-ina219-current-sensor-breakout/assembly

Just wondering, have they got their documentation wrong? 
It's just, I've never heard of 40/41 being part of the same address. This because I also use multiple sc16is752 next to each other all just one decimal apart. But it could be my understanding is totally wrong ofcourse. :) 

Will try nevertheless :)

ryanmich251, 2021-01-29:
Oops, my bad. It should work with the 0x40 and 0x41 addresses. The addresses we're working with don't include the direction bit (just everything to the left of it, the direction bit is added on by probably the i2c library). I'll look into this more.

ryanmich251, 2021-01-30:
Can't seem to reproduce this issue, wired up 2 ina219 modules (one at address 0x40 and the other at 0x41) and everything seems to work as it should (even tested with the flow you provided). One thing you could try is updating your installation of this module to v0.7.6 (you're probably on 0.7.5, just released 0.7.6 yesterday). 0.7.6 mainly just updates the i2c-bus library used by the easybotics-ina219 dependency, so if that fixes your problem it was probably a bug in i2c-bus that got fixed.

If that doesn't work, could you provide a little more information about your setup, like nodejs version, node-red version, etc.?

Jamos1988, 2021-01-31:
Oh wow thanks for taking all the time! 

One question about the update, you mean upgrading node-red-contrib-easybotics-ina219 right? So, navigating to the node-red folder and npm install the module again? Or do you have any other advice of  upgrading it?

Also, here is my information:

nodejs: v10.23.0
node-red: 1.2.7

(I also have this weird problem that my gpio pins are all stopped, maybe the versions of nodejs and node-red are also causing this, because it did work before)

Jamos1988, 2021-01-31:
Another update. When disabling and enabling flows, sometimes it takes a minute for the readings to get normal again. This was very confusing because I thought that standalone the ina219 didn't work anymore. But, I still have the problem when 2 of them are active, the readings become incorrect again. The voltages of one or the other (never both) are dropping to 0.77 instead of 12/13.

I also upgraded nodejs to the newest now. (10.23.2) but I haven't succeeded in upgrading the node with:

```
cd .signalk/red
npm install ina219
git clone https://github.com/easybotics/node-red-c...ina219.git
cd node-red-contrib-easybotics-ina219
npm install
sudo npm link
This link must be moved or copied to .signalk/red/node_modules
cp /usr/lib/node_modules/node-red-contrib-easybotics-ina219-sensor /home/pi/.signalk/red/node_modules
```

ryanmich251, 2021-02-02:
Yeah, update node-red-contrib-easybotics-ina219 (note that on npm the package is called node-red-contrib-easybotics-ina219**-sensor**).

As for updating it, you shouldn't need to clone the git repo and install that way since the latest version is on npm. Just running `npm install node-red-contrib-easybotics-ina219-sensor` inside your node red directory should do what you want. If it doesn't, you may need to edit the package.json file in your node red directory and change the version for node-red-contrib-easybotics-ina219-sensor in there to 0.7.6.

Jamos1988, 2021-02-03:
Thanks! I've done the update and rebooted, update confirmed.
However, still got the issue.
Somehow now only the one connected to 0x41 gives the wrong readings, also stand-alone.
I am beginning to think the chip is bad, I'll try to connect a spare one and report the findings.

Asier21, 2023-07-18:
Hey, I have the same problem, I have 3 Ina219 sensors and a multiplexer, the 3 sensors have different I2C addresses, individually they work fine, however when connecting the 3 stops working and give me erroneous readings, in some cases I have managed to get 2 sensors to work however when connecting the third one again they go crazy. Lastly, what are you working with? an arduino or a raspberry?