easybotics/node-red-contrib-easybotics-ina219 #14: flows.nodered.org (and Palette) still redirects to 0.7.5

charliejgallo, 2021-02-03:
seems like the 0.7.6 update did not "hit" in the flow manager in Node-red and the flows page...

[image]

here they recommend something to force refresh, 

https://discourse.nodered.org/t/node-not-updated-in-palette-manager/10354

ryanmich251, 2021-02-03:
Thanks for pointing that out, just updated it on flows.nodered.org, it should show up as version 0.7.6 now.

charliejgallo, 2021-02-03:
Thanks!! installed in a fresh Pi4 with a buster image, everything OK!

[image]

closed 2021-02-03 by charliejgallo